CroatianMeteorNetwork/RMS #494: EOFError in downloadNewMask

g7gpr, 2025-01-03:
Observed a crash caused by a network error in downloadNewMask. 

I suspect a try block would fix this problem, I'll investigate more. 

```
2025/01/01 00:37:22-INFO-DownloadMask-line:90 - files/masks exists
2025/01/01 00:37:22-INFO-DownloadMask-line:108 - Most recent flat /home/au001v/RMS_data/CapturedFiles/AU001V_20241231_115040_084674/flat.bmp
2025/01/01 00:37:22-INFO-DownloadMask-line:113 - Uploading to files/masks/ as AU001V_20241231_flat.bmp
2025/01/01 00:54:08-WARNING-DownloadMask-line:131 - Could not upload latest flat
2025/01/01 00:54:08-INFO-Logger-line:44 - Traceback (most recent call last):
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
2025/01/01 00:54:08-INFO-Logger-line:44 - return _run_code(code, main_globals, None,
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
2025/01/01 00:54:08-INFO-Logger-line:44 - exec(code, run_globals)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/source/RMS/RMS/StartCapture.py", line 941, in <module>
2025/01/01 00:54:08-INFO-Logger-line:44 - downloadNewMask(config)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/source/RMS/RMS/DownloadMask.py", line 140, in downloadNewMask
2025/01/01 00:54:08-INFO-Logger-line:44 - sftp.lstat(remote_mask)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/sftp_client.py", line 511, in lstat
2025/01/01 00:54:08-INFO-Logger-line:44 - t, msg = self._request(CMD_LSTAT, path)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/sftp_client.py", line 856, in _request
2025/01/01 00:54:08-INFO-Logger-line:44 - num = self._async_request(type(None), t, *args)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/sftp_client.py", line 881, in _async_request
2025/01/01 00:54:08-INFO-Logger-line:44 - self._send_packet(t, msg)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/sftp.py", line 209, in _send_packet
2025/01/01 00:54:08-INFO-Logger-line:44 - self._write_all(out)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/sftp.py", line 173, in _write_all
2025/01/01 00:54:08-INFO-Logger-line:44 - n = self.sock.send(out)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/channel.py", line 799, in send
2025/01/01 00:54:08-INFO-Logger-line:44 - return self._send(s, m)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/channel.py", line 1206, in _send
2025/01/01 00:54:08-INFO-Logger-line:44 - self.transport._send_user_message(m)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/transport.py", line 1967, in _send_user_message
2025/01/01 00:54:08-INFO-Logger-line:44 - self._send_message(data)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/transport.py", line 1943, in _send_message
2025/01/01 00:54:08-INFO-Logger-line:44 - self.packetizer.send_message(data)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/packet.py", line 435, in send_message
2025/01/01 00:54:08-INFO-Logger-line:44 - self.write_all(out)
2025/01/01 00:54:08-INFO-Logger-line:44 - File "/home/au001v/vRMS/lib/python3.9/site-packages/paramiko/packet.py", line 368, in write_all
2025/01/01 00:54:08-INFO-Logger-line:44 - raise EOFError()
2025/01/01 00:54:08-INFO-Logger-line:44 - EOFError
```

dvida, 2025-01-03:
Thanks, Dave, keep us posted! I checked the code and the sftp.lstat(remote_mask) command is already in a try block in the latest version of the code. Strangely, it says it's been added by you 12 months ago, could the station have been running a very old version of the code?

g7gpr, 2025-01-03:
Good thought. Will check.

g7gpr, 2025-01-03:
It appears that the last successful run was up to date. I'll spend some time looking at the code seeing if there is a way in that avoids the try block.

g7gpr, 2025-03-06:
Not observed again, presume resolved.

closed 2025-03-06 by g7gpr