CroatianMeteorNetwork/RMS #495: Continuous capture with long-running external script. Reboot issue

devcenary, 2025-01-03:
If continuous capture is enabled, start_time is set to True:
https://github.com/CroatianMeteorNetwork/RMS/blob/9b178f0a5a63a633200f776eb821fce3ec399a86/RMS/StartCapture.py#L1037

If lock file is present due to a long-running external script for instance, reboot is not retried:
https://github.com/CroatianMeteorNetwork/RMS/blob/9b178f0a5a63a633200f776eb821fce3ec399a86/RMS/StartCapture.py#L1094-L1096

dvida, 2025-01-08:
@MaadhyamRana could you look into this when you get the time?

MaadhyamRana, 2025-01-09:
^ Here is a PR for this, let me know if this is the intended behavior.

dvida, 2025-01-09:
Merged into prerelease, it will be merged into master after a bit of testing.

closed 2025-01-09 by dvida