CroatianMeteorNetwork/RMS #499: Feature request: allow popping from stash during update

devcenary, 2025-01-12:
I'm wondering if popping from stash after [pulling from repository during update](https://github.com/CroatianMeteorNetwork/RMS/blob/59b7dc14fc4b3d209607d457c6a5b4ad0f3a4a63/Scripts/RMS_Update.sh#L61-L66) can be considered. This can be controlled through env variable for instance. Something like:

```bash
STASH_POP_AFTER_UPDATE=${STASH_POP_AFTER_UPDATE:-false}
if [ "$STASH_POP_AFTER_UPDATE" = "true" ]; then
    git stash pop
    if [ $? -ne 0 ]; then
        echo "Conflicts detected. Reverting..."
        echo "Please pop from stash and resolve conflicts manually."

        # In case of conflicts pop does not remove commit from stash, so we can safely reset
        git reset --hard HEAD
    fi
fi
```

This will provide advanced users with greater flexibility to enhance RMS behavior, as the current approach is limited to only using a config file.

Cybis320, 2025-01-12:
Personnally, I like that a lot.

markmac99, 2025-01-12:
I think RMS used to do this, but i believe it caused problems. Camera operators usually do not understand git error messages and also usually don't see them because stations run unattended and headless and so stations sometimes ended up unable to start due to git conflict markers being embedded in files, and would also fail to receive code updates entirely, leading to inefficient or even incorrect results, out of date code, and excessive support workload dealing with issues.  The proposed approach would probably avoid the 1st issue, but could leave stations out of date.  I think its really important that there's as much consistency as possible across the network. 

I'm also curious as to what problem this is intended to solve. Personally I do not think any "production" system should have any bespoke changes to the RMS codebase which would need to be stashed, and you would not normally be running an unattended update on a development station. Its easy enough to manually do git stash / git stash apply before and after running the update script?

devcenary, 2025-01-12:
The proposed approach doesn't differ from the current one by default (`STASH_POP_AFTER_UPDATE` is false by default) and is supposed to be used by operators who intentionally enabled it. So, it shouldn't cause any additional problems to regular operations or leave stations out of date.

There are 2 main intents I envision:
1. Add more logging to better understand the processing flow and to debug/troubleshoot/provide data for more informed config parameters selection.
2. Test some changes in prod station by running them in shadow mode when there's no dev station available and the changes are hard to be tested by other means like external scripts.

In both cases, I'd like to run the station unattended most of the time and keep the source code updated. That said, it obviously opens up a possibility to mess up with prod flow, but it won't be hard for the operator to get the station back in sync with the repo.

closed 2025-01-14 by devcenary