CroatianMeteorNetwork/RMS #506: I can't use Skyfit2 on Ubuntu for MP4 videos

aelbazsanz, 2025-01-20:
Hello,

I have installed RMS and its utilities, including SkyFit2, several times, both on Ubuntu 22 and Ubuntu 20, following the procedure described, both with the supplied script (GRMSL_Install.sh) and with the conda packages, as documented in the project's Github page.

In both cases, always at the end of the process, when I try to open an mp4 video, I get the following error:

python3 -m Utils.SkyFit2 ./Estepa/M20240519_073311_Estepa.mp4 --config . --timebeg 20240518_224639.000000

0:00:04.023406629 49717 0x7f542c000b60 ERROR               ebmlread ebml-read.c:139:gst_ebml_peek_id_length:<matroskademux0> Invalid EBML ID size tag (0x0) at position 0 (0x0)

This same command and video works for other colleagues with an older version of Skyfit2. It also works for me on Windows.

I have installed all the dependencies that you have been requesting. I have also seen that there is another issue in which this error is discussed and that says that it is solved by changing the media_backend, but I don't know how to do it.

Thank you very much for any help you can give me.

Ángel

satmonkey, 2025-01-20:
To change the backend to opencv, put/modify in the .config this variable:
media_backend: cv2
What is the resolution of your mp4 file?

aelbazsanz, 2025-01-21:
Hello,

Thank you very much for your reply. Now, by changing the media_backend in the configuration I can open the file.

The video information I get with ffprobe is: (resolution 3840x2160 and 24.24 fps)

  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    creation_time   : 2024-05-18T21:46:39.000000Z
  Duration: 00:00:18.77, start: 0.000000, bitrate: 8286 kb/s
    Stream #0:0(und): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt470bg/bt470bg/smpte170m), 3840x2160, 8218 kb/s, 24.24 fps, 100 tbr, 1k tbn, 25 tbc (default)

But, when open the file with Skyfit2, I get this information:

Mask resolution (1280, 720) does not match the image resolution (3840, 2160). Ignoring the mask.
Using video file: ./Estepa/M20240519_073311_Estepa.mp4
FPS from video: 100.0
Total frames: 455
Frames to read: 256

The value for FPS in SkyFit2 not match with the video information. It's correct? The values in .config file are:

media_backend: cv2

uyvy_pixelformat: false

width: 3840
height: 2160
fps: 24.24 ; frames per second

Many thanks!
Ángel

satmonkey, 2025-01-23:
I have found the reason in the hardcoded pipeline - matroskademux element causes it.
Now, we have to wait until somebody replies to my comment in [449](https://github.com/CroatianMeteorNetwork/RMS/issues/449)

dvida, 2025-05-21:
Hi, did anyone check if my fix solved the problem? It should already be merged into master.
If it did, I'm be happy to close this issue.

closed 2025-05-28 by dvida