CroatianMeteorNetwork/RMS #510: Vignetting and extinction correction in predicted number of stars computation

devcenary, 2025-01-26:
When predicting the number of stars in Flux.py, as far as I understand the intent is to get true catalog magnitudes and correct for vignetting and extinction (from true to apparent magnitude):
https://github.com/CroatianMeteorNetwork/RMS/blob/2153a6750ac17921ea22ba81b5aeba9f69164f4e/Utils/Flux.py#L1535-L1540

However, `calculateMagnitudes` and `extinctionCorrectionApparentToTrue` used in `xyToRaDecPP` apply the inversed transformations (from apparent to true magnitude):
https://github.com/CroatianMeteorNetwork/RMS/blob/2153a6750ac17921ea22ba81b5aeba9f69164f4e/RMS/Astrometry/ApplyAstrometry.py#L697-L704

Seems like this erroneously brightens catalogue stars and reduces matched/predicted stars ratio.

dvida, 2025-01-28:
You're absolutely right, I also found that the function for selecting stars inside the FOV was buggy as well.
PR #511 addresses these issues.

devcenary, 2025-01-30:
Awesome, thanks Denis!

closed 2025-01-30 by devcenary