CroatianMeteorNetwork/RMS #519: Issue with the latest GRMSUpdater.sh update

devcenary, 2025-02-10:
`/\ :[0-9]/` doesn't seem to match the output of `who` (display number is [parenthesized](https://www.gnu.org/software/coreutils/manual/coreutils.html#The-who-Command) and $2 will return the terminal)
https://github.com/CroatianMeteorNetwork/RMS/blob/97dbfd9102c06c3c624ece7388c14c1a42865e61/Scripts/MultiCamLinux/GRMSUpdater.sh#L48

This might break updates in MultiCam setups.

Cybis320, 2025-02-10:
Would this work?
```bash
UserDisp=($(who | awk '$NF ~ /^\(:[0-9]+\)$/ {gsub(/[()]/,"",$NF); print $1,$NF}'))
```

devcenary, 2025-02-10:
Yes, this works for me, but the version before the change worked for me as well, not sure what would be the proper test case.

Cybis320, 2025-02-10:
I think the motivation for the change was an issue with usernames ending with a number like 'joe123'.

Cybis320, 2025-02-10:
OK, before we change anything, can you please confirm the current script in master is actually causing an issue for you or was this hypothetical? I agree with your assessment about the parenthesized number, but the script seems to be working for most.
If it is indeed causing issues, can you say more about your specific setup?
Thank you.
Luc

Cybis320, 2025-02-10:
Here are the outputs on my machine (it's working fine as is):
```bash
echo "=== who output ===" && who && \
echo -e "\n=== w -h output ===" && w -h && \
echo -e "\n=== who output parsed ===" && who | awk '/\ :[0-9]/ {printf("Found match: username=%s field2=%s full_line=\"%s\"\n", $1, $2, $0)}' && \
echo -e "\n=== Testing current script pattern ===" && \
echo "UserDisp array contents:" && UserDisp=($(who | awk '/\ :[0-9]/ {print $1,$2}')) && declare -p UserDisp && \
echo -e "\n=== Current terminal and display ===" && \
echo "DISPLAY=$DISPLAY" && \
echo "Terminal=$(tty)"
=== who output ===
ops      seat0        2025-02-09 23:39 (login screen)
ops      :1           2025-02-09 23:39 (:1)

=== w -h output ===
ops      tty2     -                23:39    4:06m  1:48   0.02s /usr/libexec/gnome-session-binary --session=ubuntu
ops      pts/7    -                02:44    2.00s  0.06s  0.01s w -h

=== who output parsed ===
Found match: username=ops field2=:1 full_line="ops      :1           2025-02-09 23:39 (:1)"

=== Testing current script pattern ===
UserDisp array contents:
declare -a UserDisp=([0]="ops" [1]=":1")

=== Current terminal and display ===
DISPLAY=
Terminal=/dev/pts/7
```

Could you report the output for:
```bash
echo "=== who output ===" && who && \
echo -e "\n=== w -h output ===" && w -h && \
echo -e "\n=== who output parsed ===" && who | awk '/\ :[0-9]/ {printf("Found match: username=%s field2=%s full_line=\"%s\"\n", $1, $2, $0)}' && \
echo -e "\n=== Testing current script pattern ===" && \
echo "UserDisp array contents:" && UserDisp=($(who | awk '/\ :[0-9]/ {print $1,$2}')) && declare -p UserDisp && \
echo -e "\n=== Current terminal and display ===" && \
echo "DISPLAY=$DISPLAY" && \
echo "Terminal=$(tty)"
```

devcenary, 2025-02-10:
So, `who` output difference explains why I see this issue and you don't. My output:

```
=== who output ===
rms      tty7         2025-02-08 18:42 (:0)

=== w -h output ===
rms      tty7     :0               Sat18   10:34m  0.00s  0.03s /usr/libexec/gnome-session-binary --systemd --systemd --session=ubuntu

=== who output parsed ===
awk: cmd. line:1: warning: regexp escape sequence `\ ' is not a known regexp operator

=== Testing current script pattern ===
UserDisp array contents:
awk: cmd. line:1: warning: regexp escape sequence `\ ' is not a known regexp operator
declare -a UserDisp=()

=== Current terminal and display ===
DISPLAY=:0
Terminal=/dev/pts/2
```

Cybis320, 2025-02-10:
ok, if we could all test this line:
```
UserDisp=($(who | awk '$NF ~ /^\(:[0-9]+\)$/ {gsub(/[()]/,"",$NF); print $1,$NF}')) && echo "User=${UserDisp[0]} Display=${UserDisp[1]}"
```

For me it correctly reports:
```
User=ops Display=:1
```

devcenary, 2025-02-10:
Yes, works for me:
```
User=rms Display=:0
```

Cybis320, 2025-02-10:
The above worked on Bookworm and Ubuntu 24.04.
And, it should work for usernames like joe123

Cybis320, 2025-02-10:
Out of curiosity, what system are you using?

devcenary, 2025-02-10:
Ubuntu 20.04

Cybis320, 2025-02-10:
I'm on GDM3 and GNOME
```
(vRMS) bolide@EliteCam01:~/source/RMS$ ps -A | grep -E "gdm|lightdm|sddm|xdm"
   1738 ?        00:00:00 gdm3
   1833 ?        00:00:00 gdm-session-wor
   1963 tty2     00:00:00 gdm-x-session
(vRMS) bolide@EliteCam01:~/source/RMS$ echo $XDG_CURRENT_DESKTOP

(vRMS) bolide@EliteCam01:~/source/RMS$ ps -e | grep -E "gnome|kde|xfce|mate|cinnamon|lxde"
     63 ?        00:00:00 kdevtmpfs
    931 ?        00:00:00 gnome-remote-de
   1932 ?        00:00:00 gnome-keyring-d
   2099 tty2     00:00:00 gnome-session-b
   2202 ?        00:00:00 gnome-session-c
   2227 ?        00:00:00 gnome-session-b
   2270 ?        00:00:20 gnome-shell
   2450 ?        00:00:00 gnome-shell-cal
  12430 ?        00:00:00 gnome-terminal-
```

devcenary, 2025-02-10:
lightdm and GNOME for me
```
~/source/RMS$ ps -A | grep -E "gdm|lightdm|sddm|xdm"
    865 ?        00:00:00 lightdm
   1392 ?        00:00:00 lightdm

~/source/RMS$ echo $XDG_CURRENT_DESKTOP
ubuntu:GNOME
```

closed 2025-02-12 by devcenary