CroatianMeteorNetwork/RMS #527: Log filename format changed

g7gpr, 2025-03-03:
Log file format has changed from

```
log_AU001A_20250221_005450.411985.log
```
to

```
log_AU001A_2025-02-22_044316.log
```

in prerelease

I *think* the multicamera Pi platform parses the log files to produce diagnostic data, so this change might break some other work.

g7gpr, 2025-03-20:
#542 might fix this

Cybis320, 2025-03-20:
Can you point me towards the code that parse the log name as even with #542 the log names will be different. There is no microseconds and the rolling log naming is different.

g7gpr, 2025-03-20:
@Cybis320 - I cannot, it is part of the image for the Raspberry Pi platform. I think it is used to drive the Conky display for number of detected stars.

Cybis320, 2025-03-20:
Argh, ok. Hopefully, it's not broken by this. Otherwise, the Conky display might have to be adapted. Given how broken the old log was, it's unlikely to have been working very accurately.

g7gpr, 2025-03-20:
Also an unknown number of operators have intricate custom scripts which parse the logs.

g7gpr, 2025-03-27:
Closing this, as it seems to have been resolved.

closed 2025-03-27 by g7gpr