CroatianMeteorNetwork/RMS #531: Nighttime capture starting during daylight hours (prerelease)

g7gpr, 2025-03-04:
On stations AU000U-V night time capture started during daylight hours

```
commit 92fb7520f4fb78fb3a1d59d18c926f63321bba45 (HEAD -> origin_prerelease, origin/prerelease)
Merge: 435db071 4cda95de
Author: Denis Vida <denis.vida@gmail.com>
Date:   Sun Mar 2 12:56:31 2025 -0500

    Merge pull request #522 from MaadhyamRana/ft-patch
    
    minor changes to space management
```

```
https://globalmeteornetwork.org/weblog/AU/AU000U/AU000U_20250303_024619_457218_detected/
```

This station is in UTC+8, so capture should be starting around 1129, for a 1929 UTC start. I feel that the following log entry might be related

```
2025/03/03 02:46:41-INFO-DeleteOldObservations-line:664 - Need 20.71 GB for next night
2025/03/03 02:46:41-INFO-CaptureModeSwitcher-line:51 - Next event is a sunset (2025-03-03 11:27:57.253931), switching to daytime mode
2025/03/03 02:46:44-INFO-StartCapture-line:1324 - Started capture mode switcher on a separate thread
```

However the thumbnails actually start at 0856 - still too early for capture

```
https://globalmeteornetwork.org/weblog//AU/AU000U/AU000U_20250303_024619_457218_detected/AU000U_20250303_024619_457218_CAPTURED_thumbs.jpg
```

g7gpr, 2025-03-06:
[log_AU000U_2025-03-04_235315-\[04_2353-to-05_2353\].log](https://github.com/user-attachments/files/19104071/log_AU000U_2025-03-04_235315-.04_2353-to-05_2353.log)

This time I have log information, which might give some insight into what is happening.

Is it possible that 

https://github.com/CroatianMeteorNetwork/RMS/blob/391278ec5f81621abf81e60bdf2e03fa6d1c016d/RMS/StartCapture.py#L127C9-L127C85

Noticing that a change of state has occurred, is incorrectly assuming that it has been a change of state from day to night, rather than from night to day?

Possibly I'm a commit behind; I'll update and keep monitoring.

Cybis320, 2025-03-06:
The camera was not ready for the first 9 hours. It could be pinged but its RTSP server was down. 
From midnight to 08:56 UTC:
```
2025/03/05 04:24:33-ERROR-BufferedCapture-line:644 - RTSP service not responding after all attempts. Last error: Unknown connection error
2025/03/05 04:24:33-ERROR-BufferedCapture-line:1006 - Camera connection failed: Unknown connection error - Please check logs for details
```
Finally at 8:56 UTC:
```
2025/03/05 08:56:46-INFO-BufferedCapture-line:603 - RTSP service ready after 43 attempts
```

Something funky with that camera...

g7gpr, 2025-03-06:
@Cybis320, the cameras are in UTC+8, and the cameras not being available is to be expected. There is no power supply to the 6 cameras until local 1700, or 0900 UTC. However that should not cause night time capture to start as soon as the cameras get powered up; although it might be contributing to the issue.

g7gpr, 2025-03-25:
Another strange capture session which seemed to run across a couple of days. I'll append the logs later. 

https://globalmeteornetwork.org/weblog/AU/AU001F/AU001F_20250322_110204_236826_detected/

dvida, 2025-04-04:
@g7gpr would you mind sending me and @Cybis320 the logs via email? If you did already, I apologize that I missed the email.

g7gpr, 2025-04-04:
Resent

closed 2025-04-08 by g7gpr