CroatianMeteorNetwork/RMS #538: StartCapture crashing shortly after detection finish (prerelease)

g7gpr, 2025-03-16:
Please find attached log snippet showing something which seemed to lead to capture stopping. 

```
2025/03/15 21:32:49-INFO-BufferedCapture-line:1715 - Block's max frame age: 0.148 seconds. Run's dropped frames: 809
2025/03/15 21:32:49-INFO-BufferedCapture-line:1825 - Created FT file /home/au001a/RMS_data/TimeFiles/2025/20250315-074/20250315-074_21/FT_AU001A_20250315_21>
2025/03/15 21:32:49-INFO-BufferedCapture-line:1544 - Grabbing a new block of 256 frames...
2025/03/15 21:32:59-INFO-DetectStarsAndMeteors-line:291 - TOTAL: 0 detected meteors.
2025/03/15 21:32:59-WARNING-Logger-line:83 - Traceback (most recent call last):
2025/03/15 21:32:59-WARNING-Logger-line:83 - File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
2025/03/15 21:32:59-WARNING-Logger-line:83 - return _run_code(code, main_globals, None,
2025/03/15 21:32:59-WARNING-Logger-line:83 - File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
2025/03/15 21:32:59-WARNING-Logger-line:83 - exec(code, run_globals)
2025/03/15 21:32:59-WARNING-Logger-line:83 - File "/home/au001a/source/RMS/RMS/StartCapture.py", line 1345, in <module>
2025/03/15 21:32:59-INFO-BufferedCapture-line:1715 - Block's max frame age: 0.150 seconds. Run's dropped frames: 809
2025/03/15 21:33:00-INFO-BufferedCapture-line:1825 - Created FT file /home/au001a/RMS_data/TimeFiles/2025/20250315-074/20250315-074_21/FT_AU001A_20250315_21>
2025/03/15 21:33:00-INFO-BufferedCapture-line:1544 - Grabbing a new block of 256 frames...
2025/03/15 21:33:00-WARNING-Logger-line:83 - night_archive_dir = runCapture(config, duration=duration, nodetect=cml_args.nodetect, \
2025/03/15 21:33:00-WARNING-Logger-line:83 - File "/home/au001a/source/RMS/RMS/StartCapture.py", line 646, in runCapture
2025/03/15 21:33:00-WARNING-Logger-line:83 - night_archive_dir, archive_name, _ = processNight(night_data_dir, config, \
2025/03/15 21:33:00-WARNING-Logger-line:83 - File "/home/au001a/source/RMS/RMS/Reprocess.py", line 198, in processNight
2025/03/15 21:33:02-WARNING-Logger-line:83 - calstars_name, ftpdetectinfo_name, ff_detected = saveDetections(detection_results, \
2025/03/15 21:33:02-WARNING-Logger-line:83 - File "/home/au001a/source/RMS/RMS/DetectStarsAndMeteors.py", line 302, in saveDetections
2025/03/15 21:33:02-WARNING-Logger-line:83 - _, x2, y2, background, intensity, fwhm = star_data
2025/03/15 21:33:02-WARNING-Logger-line:83 - ValueError
2025/03/15 21:33:02-WARNING-Logger-line:83 - :
2025/03/15 21:33:02-WARNING-Logger-line:83 - too many values to unpack (expected 6)
```

Cybis320, 2025-03-16:
Argh. The old old ExtractStarsFF always returns six items while the new one can return 9 items...

Cybis320, 2025-03-16:
The ExtractStars.py was updated a few days ago. The ExtractStarsFF function used to return 6 elements but now return 9 elements:
https://github.com/CroatianMeteorNetwork/RMS/blob/19d978bef66406006ef841767721473ba43825ae/RMS/ExtractStars.py#L348

However the SaveDetections function in DetectStarsAndMeteors.py still expects 6 elements:
https://github.com/CroatianMeteorNetwork/RMS/blob/19d978bef66406006ef841767721473ba43825ae/RMS/DetectStarsAndMeteors.py#L302

dvida, 2025-03-16:
I pushed a fix, let me know if it helps

dvida, 2025-03-17:
The second time's the charm, fixed and tested.

closed 2025-03-17 by dvida