CroatianMeteorNetwork/RMS #539: MorphCy.pyx asertion failed error (prerelease)

g7gpr, 2025-03-17:
```
2025/03/17 20:52:21-INFO-DetectStarsAndMeteors-line:90 - Detected stars: 64
2025/03/17 20:52:21-DEBUG-DetectStarsAndMeteors-line:96 - At least 20 stars, detecting meteors...
Traceback (most recent call last):
  File "/home/au000w/source/RMS/RMS/QueuedPool.py", line 361, in _workerFunc
    result = func(*all_args, **self.func_kwargs)
  File "/home/au000w/source/RMS/RMS/DetectStarsAndMeteors.py", line 99, in detectStarsAndMeteors
    meteor_list = detectMeteors(img_handle, config, flat_struct=flat_struct, dark=dark, mask=mask)
  File "/home/au000w/source/RMS/RMS/Detection.py", line 1110, in detectMeteors
    line_list = getLines(img_handle, config.k1_det, config.j1_det, config.time_slide, config.time_window_size,
  File "/home/au000w/source/RMS/RMS/Detection.py", line 521, in getLines
    img = morph.morphApply(img, [1, 2, 3, 4, 1])
  File "RMS/Routines/MorphCy.pyx", line 39, in RMS.Routines.MorphCy.morphApply
    img = close(img)
  File "RMS/Routines/MorphCy.pyx", line 243, in RMS.Routines.MorphCy.close
    img = cv2.morphologyEx(img, cv2.MORPH_CLOSE, kernel)
cv2.error: OpenCV(4.9.0) /io/opencv/modules/core/src/system.cpp:1803: error: (-215:Assertion failed) tlsSlotsSize > slotIdx in function 'getData'
```

Seeing this error on a few cameras. Not sure if it is missing packages, wrong OpenCV version.

dvida, 2025-03-18:
One working system has 4.7.0, perhaps try forcing that version. This seems to be an internal OpenCV error.

g7gpr, 2025-03-18:
Thanks will investigate this.

Cybis320, 2025-03-18:
Are there any patterns? Is it on Rpi or intel? Is openCV from an image or was it built on the machine?

g7gpr, 2025-03-18:
@Cybis320 not investigated in any detail, all good questions though.

However, I don't recall seeing this on master branch.

g7gpr, 2025-03-18:
Intel, built on machine.

g7gpr, 2026-11-06:
Problem cannot be recreated. Closing.

closed 2026-11-06 by g7gpr