CroatianMeteorNetwork/RMS #541: FramesFiles directories not gettting cleaned up

g7gpr, 2025-03-19:
```
drwxr-xr-x 13 au000d au000d 4.0K Feb  3 21:00 20250203-034
drwxr-xr-x 14 au000d au000d 4.0K Mar 19 09:21 20250204-035
drwxr-xr-x 13 au000d au000d 4.0K Feb 11 21:01 20250211-042
drwxr-xr-x 13 au000d au000d 4.0K Feb 14 21:00 20250214-045
drwxr-xr-x 13 au000d au000d 4.0K Feb 16 21:01 20250216-047
drwxr-xr-x 13 au000d au000d 4.0K Feb 18 21:00 20250218-049
drwxr-xr-x 13 au000d au000d 4.0K Feb 22 21:01 20250222-053
drwxr-xr-x 13 au000d au000d 4.0K Feb 23 21:00 20250223-054
drwxr-xr-x 13 au000d au000d 4.0K Feb 24 21:00 20250224-055
drwxr-xr-x 13 au000d au000d 4.0K Feb 25 21:01 20250225-056
drwxr-xr-x 13 au000d au000d 4.0K Feb 26 21:00 20250226-057
drwxr-xr-x 26 au000d au000d 4.0K Mar  3 23:00 20250303-062
drwxr-xr-x 26 au000d au000d 4.0K Mar  4 23:00 20250304-063
drwxr-xr-x 26 au000d au000d 4.0K Mar  5 23:00 20250305-064
drwxr-xr-x 26 au000d au000d 4.0K Mar  6 23:00 20250306-065
drwxr-xr-x 26 au000d au000d 4.0K Mar  7 23:00 20250307-066
drwxr-xr-x 15 au000d au000d 4.0K Mar  8 23:15 20250308-067
drwxr-xr-x 26 au000d au000d 4.0K Mar  9 23:01 20250309-068
drwxr-xr-x 26 au000d au000d 4.0K Mar 10 23:00 20250310-069
drwxr-xr-x 26 au000d au000d 4.0K Mar 12 23:00 20250312-071
drwxr-xr-x 26 au000d au000d 4.0K Mar 15 23:00 20250315-074
drwxr-xr-x 13 au000d au000d 4.0K Mar 16 10:00 20250316-075
drwxr-xr-x  5 au000d au000d 4.0K Mar 17 23:01 20250317-076
drwxr-xr-x 26 au000d au000d 4.0K Mar 18 23:00 20250318-077
drwxr-xr-x 13 au000d au000d 4.0K Mar 19 10:00 20250319-078
-rw-r--r--  1 au000d au000d 122K Mar 16 10:19 AU000D_20250316-075_frametimes.json
-rw-r--r--  1 au000d au000d  34K Mar 18 00:01 AU000D_20250317-076_frametimes.json
-rw-r--r--  1 au000d au000d 284K Mar 19 00:00 AU000D_20250318-077_frametimes.json
-rw-r--r--  1 au000d au000d 115K Mar 19 10:02 AU000D_20250319-078_frametimes.json
```

FramesFiles directories are not getting cleaned up. I see from the code that they are supposed to be cleaned up during post processing, but somehow this is not happening, sometimes.

Cybis320, 2025-03-19:
I think what may be happening is that the files are not getting cleaned up if the frames timelapse making fails. 

I wonder if we are running into system limits by post-processing all cameras simultaneously. In particular, I wonder if we are hitting the max files open limit. 

Something to check is whether all cameras frames timelapse making failed at the same moment (as all processes would be affected simultaneously).

A solution would be to post process each camera sequentially. Another option is to raise the max open files limit (I'm testing that option at the moment).

g7gpr, 2025-03-20:
Could it just be that the strategy of cleaning up during postprocessing is flawed?

If RMS is stopped, does postprocessing handle all the older directories when postprocessing runs on the next occasion.

Cybis320, 2025-03-20:
There's room for improvement on several levels. 
1. the timelapase generation process has excessive storage requirements:
    a. Processing full 24-hour batches (e.g. 103,680 frames across 6 cameras at 1 frame/5sec)
    b. Converting grayscale frames to color unnecessarily
    c. Creating full JPEG quality 100 copies of all frames (almost tripling storage needs)
    d. Running multiple camera processing simultaneously instead of sequentially
2. No retry mechanism or proper cleanup after failures
3. Timelapse generation occurs at the end of the night for the last midnight to midnight UTC period causing delays and date inconsistencies
4. Unreliable timestamp JSON file generation
5. Poor timestamp overlay readability on bright backgrounds
6. Lack of debug output or progress indicators during processing
7.  Frames timelapse are bundled with meteor archived files despite separate pipelines server-side

I'm proposing several changes to improve the timelapse generation process:

1. Split Processing into Smaller Batches:
    - Generate timelapses twice per 24 hours (day/night separation)
    - Process night footage as grayscale (3x smaller file sizes)
    - Process sequentially rather than simultaneously

2. Optimize Storage Usage:

    - Stream frames directly to ffmpeg instead of creating temporary copies

3. Improve Reliability:

    - Add retry mechanisms for failed processing attempts
    - Implement proper cleanup procedures
    - Generate timestamp JSON files during overlay creation for consistency

4. Enhanced Usability:

    - Add contrasting outlines to timestamp text for readability on any background
    - Provide detailed progress reporting and debug information
    - Store videos in logically named folders that match their content separate from the meteor archive

For a 6x720p MCL,  the current  timelapse generation process requires at a minimum 40 GB of free storage space. When the process complete, the videos and tar archives will take 8 GB of final storage space per 24hr (mostly taken up by the tars which can optionally be deleted - maybe they should be deleted by default?).

Streaming to ffmpeg would greatly reduce peak storage needs.

Cybis320, 2025-03-20:
OK, I tested the theory that it might have been a max files open limit issue, and it's not the case. In my case, it appears to be running out of storage during the MP4 making process. Even though there seems to be enough space (> 100 GB, 6x1080p) before and after...

Since there is no cleanup after a failure, the problem gets worse and worse until the storage is completely full of unprocessed JPEG frames, and there is nothing else for the space management routine to delete. 

This is obviously a high priority issue.

g7gpr, 2025-03-20:
I'm running machines with effectively no space constraints, more than 1TB of free space at all times.

g7gpr, 2025-04-04:
I've submitted #552 which addresses this issue.

g7gpr, 2025-04-08:
@Cybis320 I'm happy to close this now and will do during the next 24 hours unless you want to keep it open.