CroatianMeteorNetwork/RMS #546: Investigate max packet size and window size limiting for paramiko

g7gpr, 2025-03-25:
An operator has been having difficulties with transmitting data over an ADSL connection, and has asked if any work has been done on using the features in Paramiko to reduce the window size and packet size to prevent flooding consumer grade routers. 

I propose to branch and do some testing on this; however also considering that very slow data transfers tie up server sockets. 

The present solution at this station is to rsync all the bz2 files to a server on a fast internet connection, and transmit these at 1.5MBPs to the server.

nzstevem, 2025-03-25:
Hi guys, if I can help with any testing required please let me know. Steve

g7gpr, 2025-03-26:
@Cybis320 if you can create a branch that would be great!

g7gpr, 2025-03-26:
@Cybis320 config options would be preferred.

g7gpr, 2025-04-04:
@Cybis320, are you still working on this? I've got some time and can implement it if you're too committed.

g7gpr, 2025-04-11:
OK. I can test if the code has bugs, but it will need to be deployed at the specific station.

I'll report back next week.

nzstevem, 2025-04-11:
That's going to be hard because the link to the site is so weak - even anydesk stops and starts.

What about we deploy the code to my station here NZ005F. It normally uploads in seconds. If that then stretches out to hours we will know we've hit the spot.

?