CroatianMeteorNetwork/RMS #548: Upload Operating Window

nzstevem, 2025-03-28:
Request new functionality to enable the timing of RMS uploads to be controlled via .config

This request is a further attempt to mitigate the situation behind the previous issue #546 where RMS uploads are causing other applications relating to the operator's business on a fragile vdsl to time out. When this occurs he powers the rms raspi down.  

If the timing of the upload manager run could be controlled (delayed) then a schedule of operation could probably be settled to avoid conflict.

Could the upload manager process be included in the event monitor schedule and simply check each run if it was within a window define in .config and if it was do its thing???

Note this request does not replace issue #546. Envisage that both may need to be deployed.

g7gpr, 2025-04-04:
Starting work on this, and will probably submit a single PR for both this issue and #546.

g7gpr, 2025-04-10:
Still working on it.