CroatianMeteorNetwork/RMS #556: Cannot recover from stack overflow. - request for improved error handling

nzstevem, 2025-04-06:
The error below occurs when RMS tries to run and there is ZERO disk space available due to a blow-out in the capture directories. I've seen it multiple times.

The cause and resolution of this issue is not obvious to many operators.

I'm wondering if the scripts concerned can have a simple check added at the outset for a minimum amount of free disk space (say 5gb) and  if necessary simply terminate with a message to that effect, and a recommendation to prune the archive and capture and review space management settings. 

Error log:

If you need to update the RMS config file, you can do it now.
Any changes to the config file will be read only after this script is started again or the Pi is rebooted.

running build_ext
skipping '/home/pi/.pyxbld/temp.linux-armv7l-3.7/pyrex/RMS/Astrometry/CyFunctions.c' Cython extension (up-to-date)
skipping 'RMS.Astrometry.CyFunctions' extension (up-to-date)
running build_ext
skipping '/home/pi/.pyxbld/temp.linux-armv7l-3.7/pyrex/RMS/Routines/MorphCy.c' Cython extension (up-to-date)
skipping 'RMS.Routines.MorphCy' extension (up-to-date)
running build_ext
skipping '/home/pi/.pyxbld/temp.linux-armv7l-3.7/pyrex/RMS/Routines/Grouping3Dcy.c' Cython extension (up-to-date)
skipping 'RMS.Routines.Grouping3Dcy' extension (up-to-date)
running build_ext
skipping '/home/pi/.pyxbld/temp.linux-armv7l-3.7/pyrex/RMS/CompressionCy.c' Cython extension (up-to-date)
skipping 'RMS.CompressionCy' extension (up-to-date)
Loading the default config!
Creating directory: /home/pi/RMS_data
   Success: True
Creating directory: /home/pi/RMS_data/logs
   Sucess: True
Fatal Python error: Cannot recover from stack overflow.

Current thread 0xb6f5ead0 (most recent call first):
  File "/usr/lib/python3.7/logging/__init__.py", line 430 in usesTime
  File "/usr/lib/python3.7/logging/__init__.py", line 588 in usesTime
  File "/usr/lib/python3.7/logging/__init__.py", line 620 in format
  File "/usr/lib/python3.7/logging/__init__.py", line 880 in format
  File "/usr/lib/python3.7/logging/__init__.py", line 1034 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/handlers.py", line 72 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  File "/home/pi/source/RMS/RMS/Logger.py", line 44 in write
  File "/usr/lib/python3.7/logging/__init__.py", line 957 in handleError
  File "/usr/lib/python3.7/logging/__init__.py", line 1040 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 1132 in emit
  File "/usr/lib/python3.7/logging/handlers.py", line 70 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1519 in _log
  File "/usr/lib/python3.7/logging/__init__.py", line 1449 in log
  ...
/home/pi/Desktop/RMS_StartCapture.sh: line 22:  5020 Aborted                 python -m RMS.StartCapture "$@"
Press any key to continue... 
(vRMS) pi@NZ002L:~/source/RMS $

g7gpr, 2025-04-10:
Hi @nzstevem, I think that functionality is already in the master RMS branch, and is enabled by default. 

```
; Extra GB space left on the SD card nightly, accounting for the predicted
; size of all FF files.
extra_space_gb: 5
```

Maybe that allowance needs to be increased?

nzstevem, 2025-04-10:
Hi Dave,

 

I’m thinking that’s all good. But what happens after the end of night? There are other activities running before the next “start” such as other software updates and then there is what an operator might do inadvertently.

 

I was thinking that a check, just the check, for a minimum of free space would be useful at the start of each phase of the processing (including day time capture) if that then avoided the need for a manual intervention, or made the reason for trouble more obvious. 

 

We have a bunch of cameras where the operator has really limited technical knowledge or motivation to be involved at a technical level. To reduce the load on those who  do provide the support to these cameras we need to chip away at the underlying causes of issues that end up requiring an expert intervention (as they say, ctl-alt-del never really solves anything).