CroatianMeteorNetwork/RMS #562: Surface output of git.status() in observation data

nzstevem, 2025-04-13:
Enhancement

Observation data includes commit_data and commit_hash. Both these attributes reflect the status of the local client repo - the last date an update was applied to the local repo, and the local hash.

It would be useful to be able to easily determine if the latest software commit on the central server has been successfully pulled. This can be established using the "status()" command. See below.

Suggest new name "rms_status", and then a parsed version of response to status(), either "up to date", or "behind n commits".

[image]

nzstevem, 2025-04-17:
All I'm looking for is some way to easily tell if a remote site is running the latest software. The hash is difficult because you need to have something to compare it to, it's tedious to visual compare the long strings too ensure they match, and it doesn't tell you how far behind you are. I'm no git expert but I've now read up about the fetch. Fetch/status seems the way to me because it gives you a few more details. directly.

g7gpr, 2025-04-22:
Is there some reason why 

```
commit_date                     : 20250411_221611 
```
is not enough?

For the hash strings, generally just looking at the last few characters is enough. 

However, I do think it is possible to work out if the code is up to date using something like

```
git ls-remote https://github.com/CroatianMeteorNetwork/RMS master
```

Obviously, this needs translating into python.

dvida, 2025-04-22:
Just the date might not be granular enough if there were many updates in a
day.

Ideally, there would be a mechanism that would simply report whether the
version is the latest one and if not, how many commits (and/or days) it's
behind.

Cheers,
Denis

nzstevem, 2025-04-24:
> Is there some reason why
>
> ```
> commit_date                     : 20250411_221611
> ```
>
> is not enough?
>
> For the hash strings, generally just looking at the last few characters is enough.
>
> However, I [...]

If I look at the commit dates over a list of cameras they aren't all the same for the same hash. I believe the commit shown is the date the local repo is updated. And then when you see a different hash is that because it is an early adopter or a laggard? If we can spot cameras that aren't updating timely we can possibly head of future trouble while we still have remote control. Perhaps 50% of the cameras in NZ are operated by keen people without technical skills. There is a small group of people who do remote support and even visit sites to resolve issues. If we can get absolute setup consistency across this group I know from a previous life the support load will be reduced.

g7gpr, 2025-05-30:
> All I'm looking for is some way to easily tell if a remote site is running the latest software. The hash is difficult because you need to have something to compare it to, it's tedious to visual compare the long strings too ensure they match, and it doesn't tell you how far behind you are. I'm no git expert but I've now read up about the fetch. Fetch/status seems the way to go tome because it gives you a few more details. directly.

The trouble is, if git is behind, it is probably because fetch is not working; so you would be comparing a lagging local copy, using the same tool that was failing to update the local repo. It would always report that everything was up to date, even if it was not. 

This is on my list of things to look at; as soon as the testing work on the next release is done. I have a strategy that works, but it needs making robust.

g7gpr, 2025-06-01:
OK. I'm starting work on this. It's an interesting project, but for the next round of merge from prerelease - probably 2025Q4

apevec, 2025-06-01:
> you would be comparing a lagging local copy

`git ls-remote https://github.com/CroatianMeteorNetwork/RMS master`
does not depend on a local copy, in fact you can ran it anywhere outside git checkout directory

g7gpr, 2025-06-01:
AFAIK, and happy to be educated here, 

```
git ls-remote https://github.com/CroatianMeteorNetwork/RMS master
```

will only give me the latest commit on branches; please correct me if I'm wrong.

However, 

```
cd ~/tmp/
mkdir CommitHistory
cd CommitHIstory
git clone https://github.com/CroatianMeteorNetwork/RMS --filter=blob:none --no-checkout
```

Will download the whole commit history, on all branches, but no file contents, and allow me to search for the time and date of a commit, across all commits on all branches using 

```
git show -s --format=%ci 6b110c5
```

g7gpr, 2025-06-01:
I'm struggling with local repos configured with multiple remotes.

g7gpr, 2025-06-01:
However 

```
git fetch --filter=blob:none g7gpr
```

Will fetch the commit history, but not the actual files of my configued remote. This is a solution; but unfortunately, I'm going to have to change my "remote-url" from being a single variable, to being a list of urls.

g7gpr, 2025-06-02:
I've got a prototype up and running to demonstrate that all the functions work. 

Some local development work in in progress, so our latest local commit is ahead of the remote

```
python -m Utils.RepoCheck
Local commit 6cf45152fd8493cec95eaf00aada31edbce027c4 on branch feature/commit_behind_counter with date 2025-06-02 11:15:54+08:00 not found on remote
```

Everything is up to date

```
python -m Utils.RepoCheck

Current local commit is 2ac8e2290b432ed6a6d8844613968d92d7bc8a47 on branch feature/commit_behind_counter at 2025-06-02 11:34:36+08:00
Latest remote commit is 2ac8e229 on 2025-06-02 11:34:36+08:00
Local commit 2ac8e2290b432ed6a6d8844613968d92d7bc8a47 on branch feature/commit_behind_counter is
0.00 hours ahead of remote commit 2ac8e229 on remotes/g7gpr/feature/commit_behind_counter
Local repository is up to date with remote repository
```

There is some problem...

```
(vRMS) gmn@pemberton:~/source/RMS$ python -m Utils.RepoCheck

Current local commit is e67290dd55b2f9c7f67b2815b9f20792df117ece on branch g7gpr_feature_var_star_report at 2024-09-02 09:29:50+00:00
Latest remote commit is a05b4fa9 on 2024-09-16 02:21:20+00:00
Local commit e67290dd55b2f9c7f67b2815b9f20792df117ece on branch g7gpr_feature_var_star_report is
328.86 hours behind
Local is more than 36 hours behind, something might be wrong.
```

Fix it up

```
(vRMS) gmn@pemberton:~/source/RMS$ Scripts/RMS_Update.sh 
(vRMS) gmn@pemberton:~/source/RMS$ python -m Utils.RepoCheck
Current local commit is a05b4fa9bfaec817c506a71e81ff959c540ac132 on branch g7gpr_feature_var_star_report at 2024-09-16 02:21:20+00:00
Latest remote commit is a05b4fa9 on 2024-09-16 02:21:20+00:00
Local commit a05b4fa9bfaec817c506a71e81ff959c540ac132 on branch g7gpr_feature_var_star_report is
0.00 hours ahead of remote commit a05b4fa9 on remotes/g7gpr/feature/var_star_report
Local repository is up to date with remote repository
```

That's the basic functionality.

Fix it up

```

nzstevem, 2025-06-02:
That looks great to me. The only thing I'd suggest is that when this surfaces in the observation report perhaps days rather than hours is more relatable (especially when the interval is in months as some of them are).  But that is a minor detail.  Nice.

g7gpr, 2025-06-26:
```
StationID                                 : AU001A
commit_date                               : 20250624_225927
commit_hash                               : 038934ba63e47817eea3db1029189302701beeb9
repository_lag_remote_days                : 0.285
```

the tag is repository_lag_remote_days

This should be available on AU001A to AU001Z in a few hours.

g7gpr, 2025-08-09:
Closing this now, I think this work is done.

closed 2025-08-09 by g7gpr