CroatianMeteorNetwork/RMS #609: [prerelease] Utils.MigrateConfig removes settings from .config file

g7gpr, 2025-05-30:
Config file before migration contains the following settings

```
[FireballDetection]

; Flag for enabling/disabling fireball detection
enable_fireball_detection: true
; Subsample to 16x16 squares (default 16)
subsampling_size: 16
; Weight for stddev in thresholding for fireball extraction
k1: 7.0
; Absolute offset in thresholding for fireball extraction
j1: 10
```

and

```
[MeteorDetection]

; Minimum number of stars required in order to run the detection
ff_min_stars = 20

; Binning (only supported for videos, images, and vid files, but no FF files!)
; -------
; Bin images before detection (has to be a factor of 2, e.g. 2, 4, 8, etc.). The X, Y coordinates in
; detections will be rescaled to the original size. 1 = do not bin.
detection_binning_factor: 1
; The image can be binned by either averaging ('avg') or summing ('sum') pixel intensities in the bin window.
; Note that the output image data type is uint16, so be careful of integer overflows when using "sum" method!
detection_binning_method: avg

; Thresholding and KHT parameters
; -------------------------------
; Weight for stddev in thresholding for faint meteor detection
k1: 3.5
; Absolute levels above average in thresholding for faint meteor detection
j1: 12
; Maximum ratio of white to all pixels on a thresholded image (used to avoid searching on very messed up
; images)
```

after migration 

```
[FireballDetection]

; Flag for enabling/disabling fireball detection
enable_fireball_detection: true

; Subsample to 16x16 squares (default 16)
subsampling_size: 16

; Weight for stddev in thresholding for fireball extraction
k1: 7.0

; Absolute offset in thresholding for fireball extraction
j1: 10
```

but k1 is missing from the next section

```
[MeteorDetection]

; Minimum number of stars required in order to run the detection
ff_min_stars: 20

; Binning (only supported for videos, images, and vid files, but no FF files!)
; -------
; Bin images before detection (has to be a factor of 2, e.g. 2, 4, 8, etc.).
; The X, Y coordinates in detections will be rescaled to the original size.
; 1 = do not bin.
detection_binning_factor: 1

; The image can be binned by either averaging ('avg') or summing ('sum') pixel
; intensities in the bin window. Note that the output image data type is uint16,
; so be careful of integer overflows when using "sum" method!
detection_binning_method: avg

; Thresholding and KHT parameters
; ----
; Absolute levels above average in thresholding for faint meteor detection
j1: 12

```

g7gpr, 2025-05-30:
I've just noticed that something strange has happened to the underlining, 

Before 

```
; Thresholding and KHT parameters
; -------------------------------

; Weight for stddev in thresholding for faint meteor detection
k1: 3.5

; Absolute levels above average in thresholding for faint meteor detection
j1: 12
```

After

```
; Thresholding and KHT parameters
; ----
; Absolute levels above average in thresholding for faint meteor detection
j1: 12
```

Almost like the line length is being applied to the underline. I don't know, I can't really follow the logic here.

Cybis320, 2025-05-30:
I wonder if this is a leftover from not recognizing sections with comments, which should be fixed now. Any chance you could try again from the original backup copy? I can't reproduce (yet).

Cybis320, 2025-05-30:
Could you give me the complete offending config file please?

g7gpr, 2025-05-30:
Yes.. two minutes

g7gpr, 2025-05-30:
[config_initial.txt](https://github.com/user-attachments/files/20526624/config_initial.txt)

Strangely, it works on Python 3.11.2, but not 3.9.2. I wouldn't believe it, except I'm seeing it with my own eyes.

g7gpr, 2025-05-31:
The output buffer never get flushed; adding 

```
newfile.flush()
```
after each newfile.write() fixes the problem for all python versions I can test. 

I can submit a PR, but I think others who are more familiar with the code should do this.

closed 2025-05-31 by dvida