CroatianMeteorNetwork/RMS #622: [prerelease] Running RMS.BufferedCapture from the command line causes seg fault

g7gpr, 2025-06-06:
```
(vRMS) au000a@walnut:~/source/RMS$ python -m RMS.BufferedCapture
Popen(['git', 'version'], cwd=/home/au000a/source/RMS, universal_newlines=False, shell=None, istream=None)
Popen(['git', 'version'], cwd=/home/au000a/source/RMS, universal_newlines=False, shell=None, istream=None)
Using fontManager instance from /home/au000a/.cache/matplotlib/fontlist-v330.json
Loaded backend agg version unknown.
Loaded backend agg version unknown.
running build_ext
skipping '/home/au000a/.pyxbld/temp.linux-x86_64-3.7/pyrex/RMS/Astrometry/CyFunctions.c' Cython extension (up-to-date)
skipping 'RMS.Astrometry.CyFunctions' extension (up-to-date)
Loading the default config!
Station code: AU000A
Media backend: gst
Creating directory: /home/au000a/RMS_data
   Success: True
Creating directory: /home/au000a/RMS_data/logs
   Success: True
2025/06/06 00:13:22-DEBUG-Logger-line:310 - Log listener configured. Current file: /home/au000a/RMS_data/logs/log_AU000A_20250606_001321.log
2025/06/06 00:13:21-DEBUG-Logger-line:394 - initLogging completed; queue listener started.
2025/06/06 00:13:22-DEBUG-BufferedCapture-line:983 - GStreamer pipeline string: rtspsrc name=src buffer-mode=1 protocols=tcp tcp-timeout=5000000 retry=5 location="rtsp://192.168.2.11:554/user=admin&password=&channel=1&stream=0.sdp/" ! rtph264depay ! h264parse ! tee name=t t. ! queue ! decodebin ! queue leaky=downstream max-size-buffers=100 max-size-bytes=0 max-size-time=0 ! videoconvert ! video/x-raw,format=BGR ! queue max-size-buffers=100 max-size-bytes=0 max-size-time=0 ! appsink max-buffers=100 drop=true sync=0 name=appsink 
2025/06/06 00:13:22-INFO-BufferedCapture-line:988 - Attempt 1: transitioning Pipeline to PLAYING state.

(process:16116): GStreamer-CRITICAL **: 00:13:22.759: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed

(process:16116): GStreamer-CRITICAL **: 00:13:22.979: gst_element_factory_make: assertion 'gst_is_initialized ()' failed
Segmentation fault
```

Cybis320, 2025-06-06:
If it happens to work feel free to submit a PR. Ty.

Cybis320, 2025-06-10:
Is this fixed?