CroatianMeteorNetwork/RMS #624: Suggest no more than two or three decimal places for Observation Summary data

peschman, 2025-06-07:
I would like to suggest that data written to Observation Summary be rounded to two or three decimal places. 
I doubt that any of our calculations are more accurate than that, given all the real-world limitations of the inexpensive hardware we use.

I believe this would include these items:
start_time
camera_fov_h
camera_fov_V
capture_duration_from_fits
dropped_frame_rate
duration
fits_file_shortfall
fits_files_shortfall_as_time
fits_files_from_duration
jitter_quality
time_first_fits_file
time_last_fits_file
total_expectd_fits

g7gpr, 2025-06-09:
Would you like to specify an ordering for the observation summary?

g7gpr, 2025-06-09:
```
camera_information              : Unavailable 
capture_duration_from_fits      : 0 
captured_directories            : 0 
clock_error_seconds             : 1.100 
clock_synchronized              : yes 
commit_date                     : 20250608_220152 
commit_hash                     : 6a96ecd3953905ba5046bb41bd328fc0c7f2d5ef 
duration                        : 100 
fits_file_shortfall             : 0 
fits_file_shortfall_as_time     : 0 
fits_files_from_duration        : 9.766 
hardware_version                : x86_64 
start_time                      : 2025-06-09 21:54:10 
stationID                       : XX0001 
storage_free_gb                 : 132.100 
storage_total_gb                : 190.250 
storage_used_gb                 : 48.410 
time_first_fits_file            : 0 
time_last_fits_file             : 0 
total_expected_fits             : 0 
total_fits                      : 0 
```

g7gpr, 2025-06-09:
Testing on AU002C, but I would value any feedback on ordering of this output. For example, should stationID be at the top?

dvida, 2025-06-09:
That's a great idea. I suggest:
station code
software info
hardware info
camera info
clock info
...

peschman, 2025-06-09:
This output looks nice, and accomplishes my goal. 

Yes, I agree that station_ID should be listed first, and would like to think more about the ordering of other values, if we are stepping away from alphabetical ordering.

Here's a bit more of the thinking behind my initial request (written before I saw these nice results):
To my way of thinking, there is nothing significant about the value shown in the fourth decimal place, or perhaps even the third decimal place. The issue, for me is that we are deceiving ourselves by overstating these results, they may be calculated with high precision, but they are based on very loose data to begin with. For instance, nowhere do we have timers with a level of time accuracy to seconds at six decimal places. Dropped frame rate is a fairly rough estimation to begin with. Jitter quality is a useful metric, but not with fifteen decimal places.

Granted, others may feel differently, and I'd like to hear rational for leaving things as is.

Excessive number of decimal places?
start_time seconds shown with 	six 
fov_h or fov_v				six
camera_duration_from_fits	twelve
dropped_frame_rate		fourteen
duration					six
fits_file_shortfall_as_time		six
fits_files_from_duration		twelve
jitter_quality				fifteen
time_first_fits_file			six, with last three all zeros
time_flast_fits_file			six, with last three all zeros

I'd argue that even the json version should be rounded off at a suitable point, if we can agree on what that point might be.

g7gpr, 2025-06-09:
I agree that anything after the 3 decimal point is probably not reliable info. It is just noise. However, I do think that, where practical, we should store at full precision (even if it is excessive), and the application that presents the data to the end user should do the rounding. 

This is why I think it is right to round off the .txt file, but not the .json file. 

The code is still too ugly to submit a PR, but it is at

https://github.com/g7gpr/RMS/blob/feature/observation_summary/reduce_precision/RMS/Formats/ObservationSummary.py

dvida, 2025-06-10:
Agreed, let's keep the machine readable files at max precision, while files for human consumption should be formatted.

g7gpr, 2025-06-10:
I also think we should squash any clock error which is less than 10 seconds to zero.

g7gpr, 2025-06-10:
```
stationID                       : AU002C 
commit_date                     : 20250610_114432 
commit_hash                     : a9de42c5d01cf2c8d895df2d36ec998e77e8eecc 
hardware_version                : x86_64 
captured_directories            : 23 
storage_used_gb                 : 2398.170 
storage_free_gb                 : 1080.600 
storage_total_gb                : 3665.020 
camera_lens                     : 6mm 
camera_fov_h                    : 53.690 
camera_fov_v                    : 29.978 
camera_pointing_alt             : 40.72 degrees 
camera_pointing_az              : 108.33 degrees 
camera_information              : 50H20L 
clock_error_seconds             : 1.100 
clock_synchronized              : yes 
start_time                      : 2025-06-10 11:45:40 
duration                        : 100 
time_first_fits_file            : 2025-05-27 10:20:14 
time_last_fits_file             : 2025-05-27 12:24:24 
total_expected_fits             : 728 
total_fits                      : 720 
fits_files_from_duration        : 9.759 
fits_file_shortfall             : 8 
fits_file_shortfall_as_time     : 00:01:21 
capture_duration_from_fits      : 7459.980 
```

dvida, 2025-06-10:
It looks good - how is the clock error determined and what does it mean exactly?

g7gpr, 2025-06-10:
Clock error is the difference between the machine time and a direct query on a time server.

```
    REF_TIME_1970 = 2208988800  # Reference time
    client = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    client.settimeout(5)
    data = b'\x1b' + 47 * b'\0'
    try:
        client.sendto(data, (addr, 123))
        data, address = client.recvfrom(1024)
    except socket.timeout:
        print("NTP request timed out")
        return None
    except Exception as e:
        print("NTP request failed: {}".format(e))
        return None
    if data:
        t = struct.unpack('!12I', data)[10]
        t -= REF_TIME_1970
        return t
    else:
        return None
```

This was implemented because some stations were running without any time sync at all, and it highlighted this problem, and in theory could allow any interesting observations from such a station to be recovered.

g7gpr, 2025-06-11:
If any form of time sync is working, then the measurement uncertainty in the check will dominate the response.

If time sync is not working, then the clock drift will start to dominate.

This value is intended to measure the clock drift of the local machine against a known good time source, without relying on a local service working.

It does not measure time uncertainty. Happy to implement this using the existing service tools.

dvida, 2025-06-11:
Got it. Let's try to make this value meaningful - would it make sense to cite the nominal number and then add some percentage as the uncertainty, or some fixed number? E.g. 2.3 +/- ~1 ?

dvida, 2025-06-11:
Got it. How about we try to estimate this delay?

Here's an example:

```
import socket
import struct
import time

REF_TIME_1970 = 2208988800

def get_ntp_time_raw(addr='pool.ntp.org'):
    client = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    client.settimeout(5)
    data = b'\x1b' + 47 * b'\0'
    
    try:
        t1 = time.time()
        client.sendto(data, (addr, 123))
        data, address = client.recvfrom(1024)
        t4 = time.time()
    except socket.timeout:
        print("NTP request timed out")
        return None
    except Exception as e:
        print("NTP request failed:", e)
        return None

    if data:
        unpacked = struct.unpack('!12I', data)
        t2 = unpacked[8] - REF_TIME_1970  # Receive timestamp
        t3 = unpacked[10] - REF_TIME_1970 # Transmit timestamp

        # Estimate round trip delay and offset
        delay = (t4 - t1) - (t3 - t2)
        offset = ((t2 - t1) + (t3 - t4)) / 2
        print("Estimated delay:", delay)
        print("Estimated offset:", offset)

        return t3 + offset  # Adjusted time
    return None
```

g7gpr, 2025-06-11:
```
clock_error_seconds             : 0.1 
clock_error_uncertainty         : 0.208 
clock_synchronized              : yes 
```

The code to extract fractional seconds from NTP is tricky, and needs more thought.

g7gpr, 2025-06-11:
Also report on which time client is in use.

g7gpr, 2025-06-11:
```
(vRMS) au002c@pemberton:~/source/RMS$ chronyc tracking
Reference ID    : A29FC87B (time.cloudflare.com)
Stratum         : 4
Ref time (UTC)  : Wed Jun 11 23:41:47 2025
System time     : 0.001156372 seconds fast of NTP time
Last offset     : +0.001779894 seconds
RMS offset      : 0.004776987 seconds
Frequency       : 25.772 ppm slow
Residual freq   : +0.286 ppm
Skew            : 16.986 ppm
Root delay      : 0.052910101 seconds
Root dispersion : 0.006484307 seconds
Update interval : 64.5 seconds
Leap status     : Normal
```

dvida, 2025-06-12:
That's a great idea. How about it first tries to run the chrony command to get the numbers and if it fails, it runs our simple code? And then add a parameter called "clock_measurement_source" which can either be chrony or RMS.

peschman, 2025-06-12:
We still plan to add the .config setting for protocol, right?
That's a different issue, but related to Observation Summary, but now you're adding stuff related to clock accuracy, not just number of decimal places.

Perhaps related to your test branch environment, I don't see these items being reported:
detections_after_ml             : 26 
dropped_frame_rate              : 0.0 
dropped_frames                  : 0 
jitter_quality                  : 100.0 
media_backend                   : gst 
photometry_good                 : True

g7gpr, 2025-06-12:
@peschman 

```
detections_after_ml : 26
dropped_frame_rate : 0.0
dropped_frames : 0
jitter_quality : 100.0
media_backend : gst
photometry_good : True
```

These are missing because I am doing my testing during the night when the observation is in progress.

Yes, the other two issues are addressed by #628 and #621. These are waiting to be merged in. 

@dvida 

Yes, add clock_measurement_source, NTP / Chrony / etc, if possible get the clock error and uncertainty from the time client, if not then run our own code. 

I'll take a look at this work tonight.

g7gpr, 2025-06-15:
Starting work on discerning which time sync client is in use.

g7gpr, 2025-06-15:
output when chronyc installed, but not synchronized

```
time_client                     : chrony 
clock_ahead_ms                  : Unknown 
clock_error_uncertainty_ms      : Unknown 
clock_synchronized              : False 
```

installed and synchronising

```
time_client                     : chrony 
clock_ahead_ms                  : -0.01 
clock_error_uncertainty_ms      : 51383.4 
clock_synchronized              : True 
```

Note very high uncertainty caused by initial root dispersion

```
david@lap-deb-01:~$ chronyc tracking
Reference ID    : A29FC801 (time.cloudflare.com)
Stratum         : 4
Ref time (UTC)  : Sun Jun 15 03:49:54 2025
System time     : 0.000010589 seconds slow of NTP time
Last offset     : +0.003550051 seconds
RMS offset      : 0.003550051 seconds
Frequency       : 3.696 ppm fast
Residual freq   : +468.760 ppm
Skew            : 1000000.000 ppm
Root delay      : 0.009203561 seconds
Root dispersion : 47.988433838 seconds
Update interval : 1.2 seconds
Leap status     : Normal
```

After a few control loop iterations

```
camera_information              : Unavailable 
time_client                     : chrony 
clock_ahead_ms                  : -2.146 
clock_error_uncertainty_ms      : 20.305 
clock_synchronized              : True 
```

g7gpr, 2025-06-15:
Now NTP, firstly using ntpstat, which is not always installed. 

```
time_client                     : ntp 
clock_ahead_ms                  : Unknown 
clock_error_uncertainty_ms      : Unknown 
clock_synchronized              : Unknown 
```

After time to synchronize

```
time_client                     : ntp 
clock_ahead_ms                  : Unknown 
clock_error_uncertainty_ms      : 33 
clock_synchronized              : True 
```

Remove ntpstat and parse output of 

```
ntpq -pw
```

```
time_client                     : ntp 
clock_ahead_ms                  : Unknown 
clock_error_uncertainty_ms      : 59.547 
clock_synchronized              : True 
```

g7gpr, 2025-06-15:
Using ntp, without ntpstat installed, and not synchronized with no internet connection

```
time_client                     : ntp 
clock_ahead_ms                  : Unknown 
clock_error_uncertainty_ms      : Unknown 
clock_synchronized              : Unknown 
```

g7gpr, 2025-06-15:
Should I merge pr #628 and #621 into this pr?

dvida, 2025-06-15:
Yes please, merge all obs_summary related work into one PR. Let me know once it's ready for merging.

g7gpr, 2025-06-15:
Added in, but need to test for a couple of nights before ready for merging.

g7gpr, 2025-06-24:
This work is done, closing for housekeeping.

closed 2025-06-24 by g7gpr