CroatianMeteorNetwork/RMS #631: Check Platepar  / Config at start of processing, to warn operators

g7gpr, 2025-06-08:
```
2025/06/08 13:28:41-INFO-Reprocess-line:132 - The FOV in the platepar is not within a factor of 2 of the FOV in the config file! Not using the platepar...
```

This warning appears after an observation session, and fit is not performed, even if a suitable platepar is available.

Maybe this check could be run when RMS Starts, so that an operator could be warned at the start of the session.

Cybis320, 2025-06-08:
If it's happening irregularly, I think the filter is doing its job weeding out bad recalibration. The original pp is probably fine but the recalibration has produced a bad solution.

g7gpr, 2025-06-08:
I replaced the original .config with one from an adjacent camera, changed the stationID, but did not update the ,config FOV settings for this specific camera.