CroatianMeteorNetwork/RMS #643: `frame_days_to_keep` only delete processed files but not the raw images

Cybis320, 2025-06-10:
Currently, `frame_days_to_keep` only deletes processed files but not raw images. I think it was deemed sufficient because those images were supposed to be archived or deleted. Currently, they are supposed to be deleted after the code is done generating timelapse from them. However, there are clearly cases where this isn't happening (for various reasons.)

Ideally, frame_days_to_keep should delete the specified number of days by deleting relevant `<hour>` directories but always preserve the last 24 hr to produce timelapses.

g7gpr, 2025-06-10:
Sorry Cybis320, I'm not familiar with the code which processes frame_days_to_keep, or the directory structure for the raw images. The quota management system obliterated everything before a cut off date, and was intended as a backstop. 

I can keep working on this, but I suspect the original author of this section of code would resolve it much more efficiently.

Cybis320, 2025-06-10:
Oops, I wrongly assumed it was you. I take a dig at it. And yes, quota management should do the job nicely, but if it's disabled, I think it makes sense for frame_days_to_keep to handle this too - I've never verified QM on this case.

g7gpr, 2025-06-10:
QM was extended to cover this exact situation, which occurred a few times during the development work, so QM makes a list of all the files in all the directories, sorts by time, and obliterates the oldest until under quota. It's not very intelligent, but since the development work has been completed, I don't recall seeing it operate.