CroatianMeteorNetwork/RMS #644: Scripts/RMS_Backup.sh is broken

peschman, 2025-06-11:
Scripts/RMS_Backup.sh only backs up the private and public ssh keys and the .config file.
It no longer backs up mask and platepar.
I don't know when it broke, because I had not tried to use it in quite a while.

g7gpr, 2025-06-11:
Yes, it's broken. 

The old versions of the .config file had the platepar path, the new versions do not. 

Can you test something please? Add the path to your platepar file 

```
platepar_name: ~/source/RMS/platepar_cmn2010.cal
```

Or wherever it is that you prefer to keep it, and see if the script works again?

peschman, 2025-06-11:
Yes, that addition to .config brought in the platepar, but not the mask

peschman, 2025-06-11:
Adding 
mask: mask.bmp 
to .config brings in the mask

dvida, 2025-06-11:
We removed these as some operators would hardcode paths and change the names, which would break things down the line. They should all be called "mask.bmp" and "platepar_cmn2010.cal".
However, they can either be in ~/source/RMS or ~/source/Stations/STATIONCODE/

dvida, 2025-06-11:
I suggest you hardcode the names in the script.

g7gpr, 2025-06-11:
Right. I'll get onto it.

g7gpr, 2025-06-13:
Where would you put all the backups?

g7gpr, 2025-06-14:
One small comment -j, or --junk_paths needs to go at the end if you really want to ditch the directory structure.

peschman, 2025-06-14:
Yes, I'm ignoring the .ssh setting in .config because I want the entire default .ssh directory. I think that anyone who puts keys someplace else should know enough to back them up some other way. Once the zip is built, it is easy to add more content to it.

When I tested an earlier version, I learned that the zip command will fail if there are duplicate filenames, which is exactly what you get with both single camera backup plus content from the Stations sub directories. For that reason, the zip file needs to contain the directory structure, and I think it is more clear when the directory structure is backed up anyway.

The original version backed everything up to the root of the zip, but that's not feasible when it is extended to include single camera and MultiCam duplicating filenames for .config, mask and platepar. Also, we are backing up all the MultiCam directory content, including any other files that might be there, like .config backup versions, old and new mask files, and so forth.

If you can provide instructions here or by email I will turn it into a PR, or if it is not easy to tell me how, you could just do it on my behalf. Thanks.

g7gpr, 2025-06-14:
OK. I think you should get rid of the -j flag. You want to preserve the directory structure, so that flag shouldn't be there.

It doesn't do anything where it is, but if the implementation of zip is changed, it might start to do unwelcome things in the future.

I'll put together a PR in 24 hours with whatever the latest version is that you send me.

peschman, 2025-06-14:
I don't see the -j in the new version (although --junk-paths was in the original version that is now broken), the line in the new version reads:

zip -q -r $BACKUP_FILE $CONFIG_FILE $MASK_$PLATEPAR_FILE FILE $RSA $STATIONS

This is the same file I attached earlier:
[RMS_Backup.txt](https://github.com/user-attachments/files/20741211/RMS_Backup.txt)

That attached file needs to be renamed from .txt to .sh

Where are you finding  the  -j  flag?

g7gpr, 2025-06-15:
Sorry, clearly I made a mistake. I'll put together a  PR.

closed 2025-06-16 by g7gpr

peschman, 2025-06-24:
It does not look like the new version is being distributed by RMS_Update

$ git log -1
commit 0352ab7c7b6cd280930d4e3c1fa15c349e5bf048 (HEAD -> master, origin/master, origin/HEAD)
Author: Denis Vida <denis.vida@gmail.com>
Date:   Mon Jun 23 10:13:24 2025 -0400

    Cleanup: Remove Cython build directory removal from update script

$ ls -l Scripts/RMS_Backup.sh 
-rwxr-xr-x 1 pi pi 1590 Dec 24  2022 Scripts/RMS_Backup.sh