CroatianMeteorNetwork/RMS #665: Change to handling of start delay causes all processing to start at the same time - I think

g7gpr, 2025-06-24:
https://github.com/CroatianMeteorNetwork/RMS/blob/ac401bbbf0ca06d05440739a9e0e4657204b84e6/RMS/CaptureModeSwitcher.py#L110

This change means that the duration of capture is reduced by the length of the start delay. Consequently, all the captures now end at the same time, which has caused problems on Pi5 if multiple timelapse generation tasks start at the same time. 

Is there any other workaround available to stagger processing start?

Cybis320, 2025-06-25:
Argh. I'll take a look at it tonight.

g7gpr, 2025-06-25:
https://globalmeteornetwork.groups.io/g/main/message/14578

This message might be relevant, op cured with increasing swap file. Perhaps that is the correct solution to this problem.

Cybis320, 2025-06-28:
Looking at it closer, I don't think the change causes all reprocess to start at the same time. The start and end of night should be correctly staggered as long as each camera config file contains a different delay value.

Even though the wait ends the loop at the same time for each camera, a variable wait is present a the start of the loop which causes each transition to be correctly staggered. 
The log message doesn't reflect this additional delay - maybe it should.

Additionally, continuous mode is not compatible with the random delay feature as it needs a deterministic transition time.