CroatianMeteorNetwork/RMS #692: No output from CameraControl

markmac99, 2025-07-28:
On 28/07 R Bassom reported that the CameraControl module was broken and not producing any output. Investigation revealed that this seems to be an unintended side effect of changes to the logger which now suppresses console output by default. I've raised a PR to fix this for CameraControl but other standalone modules are likely to be affected.

g7gpr, 2025-07-29:
Is everything getting printed and logged now? I'm seeing some strange things on master.

dvida, 2025-07-29:
Hmm, it's only supposed to be for CameraControl. I see the issue, I undid the PR for now. Let's think about this one a bit more.

markmac99, 2025-07-30:
@Cybis320  ref your first, it works corretly only if you are in ~/source/RMS when you run it. It malfunctions for multicam and if you run it anywhere outside ~/source/RMS on Pi. As you will be aware, when you pass the -c parameter to other modules, they run correctly, reading the config file as needed, and in any events this is a breaking change so should be fixed. 

Ref your 2nd, the problem is that the end user has no way to know if its worked because there's no output on screen!  

Ref your last, i  believe you added logging on 15/April, presumably to get some logging output when called from SwitchCameraModes(). I agree its not necessary in this module when called standalone.

dvida, 2025-07-31:
Thank you for the detailed breakdown - do you want me to remove those two lines then? I'm ok with that.

Cybis320, 2025-07-31:
I still don't understand the issue. What's not working and what's breaking exactly? What exact command used to work and now doesn’t, and what is the output?

Here's what I'm seeing on my side:
```
(vRMS) bolide@EliteCam01:~/source/RMS$ python -m Utils.CameraControl SetParam Camera DayNightColor 1 -c ~/source/Stations/US005B/.config
Loading config file: /home/bolide/source/Stations/US005B/.config
Creating directory: /home/bolide/RMS_data/US005B
   Success: True
Creating directory: /home/bolide/RMS_data/US005B/logs
   Success: True
2025/07/31 22:29:24-INFO-CameraControl-line:515 - Set Camera.Param.[0].DayNightColor to 0x00000001
Shutting down logging...
Logging shutdown complete.
```

Cybis320, 2025-08-01:
Mark, the unnecessary logging is because you added logging in the main. Your code comment says it was to silence DVRip output. DVRip output seems to be silenced without having to create a log in the main.

I see the issue is present in master and prerelease but not in alpha1 (I created a branch with most of the current fixes rolled in). I think the issue is that the logger closes early and fails to record the last outputs. alpha1 incorporates PR #684 which fixes that.

It has nothing to do with the working directory.

The immediate fix for master would be to delete the log creation in main.

Here's the output from alpha1:

```
(vRMS) ops@EliteCam01:~/source/Stations/US005B$ python -m  Utils.CameraControl GetHostname
Loading the default config!
Creating directory: /home/ops/RMS_data/US005B
   Success: True
Creating directory: /home/ops/RMS_data/US005B/logs
   Success: True
Shutting down logging...
2025/08/01 16:32:46-INFO-CameraControl-line:702 - LocalHost
Logging shutdown complete.
```

markmac99, 2025-08-01:
@Cybis320  I'm a bit unsure what your point is.  My point was that last week, CameraControl worked properly. This week, it doesn't. I didn't change anything in this module recently and its kind of annoying that I'm being blamed for this bug.

Regarding logging, when called from other RMS code, because we are or were by default logging DEBUG messages, and because the logger had been changed to seize control of every single logger stream instantiated by any code called by RMS, DVRip's debug functionality was being triggered, leading to it including passwords and SSH keys from connections in logfiles. I sincerely hope we agree that was bad. So firstly if we could stop logging every darn debug message that'd be great. Equally if this problem has been fixed, and DVRip#s messages are no longer being logged, that too would be great. 

So TLDR, if you think you have a fix, please apply it. Happy to test.

markmac99, 2025-08-01:
ps - I've unsubscribed from this topic, as i can offer no further input.

markmac99, 2025-08-01:
Reopened because the bug still exists. I appreciate this has become a heated topic, but this is an actual bug which we need to address. I appreciate Luc doesn't have time/inclination to fix it so i will do it.  

Just to reiterate, you can replicate it as follows
Open a terminal window on a station and run this 
```
(vRMS) rms@uk002f:~/source/RMS $ cd $HOME
(vRMS) rms@uk002f:~ $ python -m Utils.CameraControl -c ~/source/RMS/.config GetCameraParams
Loading config file: /home/rms/source/RMS/.config
Creating directory: /home/rms/RMS_data
   Success: True
Creating directory: /home/rms/RMS_data/logs
   Success: True
(vRMS) rms@uk002f:~ $
```
What should happen is that the camera parameters are printed to screen.  If you're in ~/source/RMS then the functionality behaves correctly, so clearly there's some hidden dependency on the source code.

markmac99, 2025-08-02:
understood. I'm currently working on a couple of other things in CameraControl so i will include this fix.  I'll build some test cases so i can test on single-station, multicam and Windows. I don't have MacOS so can't test there.

markmac99, 2025-08-02:
ps good to know that the issue of running from site-packages will be solved. This causes any amount of unexpected behaviour ! 

pps - i am going to look at building a pypi package for RMS, should be an interesting challenge but might offer some options wrt stability.

markmac99, 2025-08-02:
ppps "If only you knew the amount of time I spent on these things..." 
I expect i do :) I spent every evening for two years building the ukmon data capture, data processing, alerting, website etc, single handed and since retiring from my day job, the amount of time i spend on opensource projects and support has only increaased. We end up putting our souls into this stuff.

Cybis320, 2025-08-02:
> pps - i am going to look at building a pypi package for RMS, should be an interesting challenge but might offer some options wrt stability.

Excellent, please look at alpha1 which already uses pyproject.toml - the toml is really the minimum necessary to get going at this time. It would need additional information to be PyPi ready. RMS has dependencies that PyPi doesn't support (imreg_dft), has binary extensions (kit_module), Python extensions, etc. It would be a fun and challenging project!

markmac99, 2025-12-31:
I think this is resolved by changes made to the logger, closing it for now

closed 2025-12-31 by markmac99