CroatianMeteorNetwork/RMS #698: Picking correct camera_settings.json in multicam situation

markmac99, 2025-08-01:
I am currently testing a migration to multicam linux and hit an interesting problem.
During migration I set RMS to switch camera modes automatically. This worked fine for UK0006. 
However, the standard settings are suboptimal for UK000F and as a result, the sensor was misconfigured leading to very low star counts and no detected meteors, even when the other three colocated cameras were picking up 100+ stars and 60 meteors each.  

I realised that RMS defaults to loading ~/source/RMS/camera_settings.json - even in multi cam situations, unless a new configuation parameter camera_settings_path is added. However, i don't think this is explained anywhere in the migration notes, and so when migrating from a single-cam setup, you would not realise this.

I therefore have a couple of suggestions
1)  RMS should default to looking for camera_settings in the same folder as the .config file similar to how it looks for the platepar and mask files. After all, this is a camera-specific configuration file. If this fails, it could then look in ~/source/RMS

2) the multicam migration notes should copy the default camera_settings file to the Stations/XXnnnn folder and also warn the user they may need to tweak it. 

If this makes sense i can work on it.

markmac99, 2025-08-02:
I think this can best be done by amending ConfigReader. At present, if the camera_settings_path attribute is not present, then a value of './camera_settings.json' is hardcoded. I propose to change this as shown below:
```
   # if the section exists, and points to a valid file, use it
    if parser.has_option(section, "camera_settings_path") and os.path.isfile(parser.get(section, "camera_settings_path")):
        config.camera_settings_path = parser.get(section, "camera_settings_path")
    else:
        # check the folder that the config file was loaded from
        if os.path.isfile(os.path.expanduser(os.path.join(config.config_file_path,'camera_settings.json'))):
            config.camera_settings_path = os.path.expanduser(os.path.join(config.config_file_path,'camera_settings.json'))
        else:    
            # and finally fall back to the RMS source dir
            config.camera_settings_path = './camera_settings.json'
```
This is still hardcoding the final default value but otherwise i think it does the trick.

markmac99, 2025-08-02:
Yes, thats supported by the changes i'm making. I wanted to make sure that operators can tweak the settings if they need to, as for example i've had to do, but of course if they're lucky enough to have a set of sensors that behave the same, then its possible to share a file. 
By the way, I suspect some stations would benefit from some individualisation. i can see by looking at the ukmon livestream that there are a wide variety of contrasts between even colocated cameras such as my four. Of course, some of this is due to local sky quality and pointing direction but its too much to expect homogeneity from a mass produced low cost chip :)

Cybis320, 2025-08-02:
Yes, I should look at my station as well - although I'm unsure how to go about it. Is there a guide somewhere on how to tune gain and brightness? I suspect those are the two main levers, right?

markmac99, 2025-08-02:
ps UK000F is set as follows:

    ["SetColor", "100,55,50,50,0,0"],
    ["SetParam", "Camera", "ElecLevel", "100"],
    ["SetParam", "Camera", "GainParam", "Gain", "70"]

Lower values of gain and eleclevel produce a very gray view with all the faint stars lost in the background. I suspect its the age or version of the sensor.

Cybis320, 2025-08-02:
Thank you. I'm finding ElecLevel 100 is generally not necessary to keep the gain constant at night and is causing the washed-out image during day and twilight. Even at ElecLevel 60, the moon doesn't seem to trigger the gain to decrease.

An experiment is to lower ElecLevel to 60 on a night shot with the full moon in the FOV. Does the image change when going from ElecLevel 100 to 60? It doesn't seem to be the case with Gain 60 on the IMX291, but I have not tried with gain 70. 

At gain 60 and ElecLevel 60, on the IMX291, the gain is the limiting factor even on the brightest astronomical night scene. It seems a fraction of the stations are running gain 60 and the other 70 depending on whether setAllCameraParams.py (70) or RMS_SetCameraParams.sh (60) was used. I didn't realize this discrepancy until recently, and the default JSON settings took the gain value from the shell script (60). At gain 70, the ElecLevel may become the limiting factor sooner and it's possible a bright moon would start slightly reducing gain if the overall brightness of the scene reaches ElecLevel 60 (although not lower than Gain 60 level would occur so probably not a big deal - especially since background star brightness are used for reference).

Also, does changing the contrast affect linearity?

markmac99, 2025-08-03:
fix proposed via PR 700

@Cybis320 the values I change between night and day are:
```
    cam.set_info("Camera.Param.[0]",{"ElecLevel":expo})
    cam.set_info("Camera.Param.[0]",{"DayNightColor":cmode})
    cam.set_info("Camera.Param.[0].GainParam",{"Gain":gain})
    cam.set_info("Camera.Param.[0].ExposureParam",{"LeastTime":minexp})
```1
At night i have eleclevel/gain set to 100/70, cmode=2, minexp=40,000 ms
In the day, i have 30/30, 1, minexp = 100ms, 
I'm not sure changing minexp actually makes a difference on the 291.
These settings have worked well for me for several years. I don't reboot the cameras after changing settings. 

Apart from my own cameras, i've worked on about fifty others deployed in the UK and I've noticed each sensor can be a little different in its response. With one sensor (no longer in use) eleclevel >40 generated huge amouts of noise and poor quality images. With another (also no longer in use) eleclevel < 60 generated a grey image with only the brightest stars visible. Local sky conditions matter a lot, and tweaking the values a little on a per-camera basis can improve star detection.

markmac99, 2025-12-29:
Fix merged with PR700, closing issue

closed 2025-12-29 by markmac99