CroatianMeteorNetwork/RMS #699: Always switch camera modes

markmac99, 2025-08-01:
At present, RMS only switches camera modes when daytime capture is enabled. 

However, unless the mode is switched, the camera view is generally saturated and useless. Indeed, its probably adding some wear and tear on the sensor by leaving it in a highly sensitive mode potentially in full sun. 

So i wonder if it would make sense to either ALWAYS switch mode, or at least to have an additional config parameter "always_switch_mode" that the user could set. 

Happy to work on either.

markmac99, 2025-08-02:
Definitely need to trap that situation. 

However, I think the system actually *should* keep trying to switch mode? Otherwise the camera would be left in the wrong mode and data would be lost or unusable. In fact, this is probably already needed to support continuous capture. 

Indeed, arguably, it might be better to block data capture until a successful mode-switch has happened. If the camera gets stuck in daytime mode, then performance is poor and star magnitude data unreliable. Take a look at the video from UK000F for the night of 01/Aug to see how being in 'daytime' mode affects night-time data capture. Its pretty easy to spot when I realised the camera was in the wrong mode and tried to correct it.   

https://www.youtube.com/watch?v=_UK1Aegr4UY

As an aside I had to iterate through a few settings to get it right as unfortunately for this sensor the default values are not correct.

g7gpr, 2025-08-02:
That's exactly what I meant, and failed to write.

g7gpr, 2025-08-02:
One other concern, some old style static memory devices had a finite number of write cycles. I don't know if that applies to these sensors.

g7gpr, 2025-08-02:
Forcing the camera to reboot at local noon, rather than any other time would be good; I'm cursed by strange settings changes when the camera reboots. I sometimes think that the settings don't get applied until the camera reboots.

markmac99, 2025-08-02:
I'll see if that can be done. The [init] section of camera_settings sets the reboot time to 15:00, but it does that *before* setting the camera's clock so it might end up rebooting at 1500 China time :/ I'll do some tests with a spare camera.

The camera clocks also drift terribly, not surprising at the price point. Mine are all configured to use NTP but if you're not using that then RMS should probably reset the camera clock each day maybe? 

ps Some settings do require a reboot but most of the device and colour settings are applied immediately. You can see this if you're watching the livestream with VLC or CMS. I noticed that the camera_settings file includes a [reboot] statement at the end, so if you use that to maintain your camera settings it'll be rebooted anyway.  I don't really want the cameras rebooted so i have created custom settings that remove the reboot and add / alter the other settings to suit my older sensors.

g7gpr, 2025-08-02:
I agree, but I had a period where the cameras would reboot at UTC 1500, which is local 2300, and after rebooting would come back with different settings. 

I've customised the camera_setting files to switch NTP off, as NTP blocks the command to set the time, and the time gets set at each transition from day mode to night mode. I set the reboot time as UTC 0400, which is my local noon. I feel this could be done automatically. I did consider setting up an NTP server just pointing at the camera NIC; but it seemed like overkill.

markmac99, 2025-08-02:
| If we decide to go that way, it should be a config param. I would argue it should default to false.

I agree, thats my preferred option. I am a strong opponent of changes that alter the default behaviour unless its absolutely essential.

In regard to the sensor in daylight, i was thinking that flooding all the wells would be more likely to lead to eventual damage, but i'm no expert on exactly how CMOS chips operate.  I don't record during the day, but i do enjoy looking at the livestream from my cameras, especially when i'm away from home. 

| Does the order matter here? It sets reboot time to 1500 internal camera clock time

I have previously noticed on some camera modules that the reboot time got adjusted when i changed the timezone, so that it occurred at the same absolute time. That is to say, if i set the reboot for 1500 local time and the out of the box TZ was China, then after reseting the TZ to UTC, the reboot was now happening at 0800. This might be firmware specific, but it might be safer to swap the order, just to be sure. 

| There is the question of why not just set auto-reboot to Never since we are already rebooting on mode changes. 

We are currently only rebooting if the operator has set the camera to do daylight capture,  However, would it be useful for RMS to trigger a camera reboot anyway just before starting data processing? This would ensure the camera was always ready. 

| I would advise against removing the reboot commands on all IMX291 or any other camera unless you have carefully validated that it is not needed.  

Thats what i did. As i am using NTP for my cameras, i removed the time management settings, which i think do require a reboot. On my four cameras I've not observed any contrast or sharpness difference before and after reboot so again this might be firmware specific. My four are pretty old sensors (oldest dates to 2017, newest to 2020) .

markmac99, 2025-08-02:
@g7gpr i have an NTP server running on a Pi and all my systems including the cameras are configured to sync to it When i have time (anaaha) I'll stick a GPS hat on it and use that to improve time sync further.  If you have a spare arduino you can also knock one up fairly quickly eg following this project https://markmcintyreastro.co.uk/resources/timekeeping/ though nowadays there are even simpler options.

g7gpr, 2025-08-03:
@Cybis320 "Does the order matter here? It sets reboot time to 1500 internal camera clock time, then set the time to host time. I don't think the order matters but I'd happy to learn something new. Unless NTP is on, then all bets are off."

I don't think we should write any code that needs the host to be set to a specific time zone.

markmac99, 2025-08-03:
> I don't think we should write any code that needs the host to be set to a specific time zone.

This is trickier than i thought. In order for the camera to reboot at a specific local time, it has to either be configured to local time, or the code has to do a conversion between the camera's TZ and the local TZ and set the reboot accordingly.   For example in David's case that means subtracting 8 hours.

However, often if not always, the Pi or Linux box is configured to UTC. So, one would need to work out the "correct" local TZ to use . Is that possible?

g7gpr, 2025-08-03:
```
def computeLocalTimeOffset():

    utc_time = datetime.datetime.now(datetime.timezone.utc).replace(tzinfo=None)
    local_time = datetime.datetime.now()

    return round((local_time - utc_time).total_seconds() / 3600,2)
```

markmac99, 2025-08-03:
> ```
> def computeLocalTimeOffset():
>     utc_time = datetime.datetime.now(datetime.timezone.utc).replace(tzinfo=None)
>     local_time = datetime.datetime.now()
>     return round((local_time - utc_time).total_seconds() / 3600,2)
> ```

This works if the station is set to local time. On my stations however, which are all set to UTC, it returns zero.  

I like your idea of working out local noon and going from there. Setting the reboot to happen at whatever local noon is would probably be good enough.

g7gpr, 2025-08-03:
Of course, this all relies on the camera being set to roughly the machine time - which I think we should do on a regular basis.

markmac99, 2025-08-03:
ps - meant to add absolutely fine to work on my code, I'm just testing my changes for PR700

g7gpr, 2025-08-03:
pr700 looks good, I've run on it on a few different versions of Debian, and no issues.

markmac99, 2025-08-03:
I think the part thats missing is that the cameras are not necessarily set to the same TZ as the station.  OOTB they're set to Bejing time. Cameracontrol's init routine sets the clock time but not the timezone - I just confirmed this by resetting one of my camera to Bejing time then running "CameraTime set".   

In fact, if you change the TZ, the camera clock is autoadjusted correctly. On the other hand if you set the time, it literally just sets the clock to whatever your computer's current time is.   So you must set the TZ first, then the clock. 

This is needed because the DVRip module literally just writes the requested integer into the firmware on the camera. So if you determine the local noon is 1300 according to the Station's TZ, but the camera is set to UTC, then it'll reboot at 1300 UTC.

So I think we need to set the TZ in the init routine, but also set TZ and clock whenever running the SetAutoReboot code.

markmac99, 2025-08-04:
> I don't fully understand. With NTP off, and say, the tz left in china. When we set time and set reboot to 1500, it will reboot at system time 1500, no?

Yes, correct. The reboot process doesn't care what TZ the camera is set to. However 1500 system time is not necessarily 1500 local time. If the Pi is set to UTC, but is physically located in Denver, Colorado (UTC+7), the camera's clock will read 1500, but it is actually 1000 local time.  

I *think* Dave's change will resolve this by adjusting the reboot time based on longitude, but we should test it carefully with a range of longitudes and Pis set to different TZs, just in case there are occasions where someone has changed the TZ from UTC to their local timezone. 

> If we subsequently change the camera tz without also syncing time then yes the reboot will be something else than 1500 system time. The key is to sync time after a tz change if any.

I agree, if we change TZ we must also reset the camera time. 

It might be lot simpler to disable autoreboot, set the camera time and then reboot it at the end of data processing each day. That way we would not need to fiddle with TZs or do any complex calculations. On the other hand we'd need to consider that not all RMS stations use these cameras, and so the commands would not work for some hardware.

g7gpr, 2025-08-04:
Hi @markmac99, I'm no longer using the system time. I read the camera time, compute when station noon, by longitude, will be in camera time, and set the reboot in camera time, whatever that will be. This gets rid of any dependency on the computer's time zone.

g7gpr, 2025-08-04:
Next is to write a bit of test code to test the routine at every time zone, and every 5 degree longitude step.

g7gpr, 2025-08-19:
@markmac99 - Does the command to change the timezone work? Whenever I set a different timezone, and try to read it back, it has not changed. 

It's almost certainly my fault, but this has been holding me up for a couple of weeks.

g7gpr, 2025-08-19:
OK. I'll park this for now. Thanks for digging into it.