CroatianMeteorNetwork/RMS #703: Capture stopping on prerelease on Pi5

g7gpr, 2025-08-07:
```
2025/08/05 11:23:52-DEBUG-VideoExtraction-line:285 - [AU000L_20250805_112338_303_0090368] time for thresholding and subsampling: 0.17669129371643066s
2025/08/05 11:23:52-INFO-VideoExtraction-line:289 - Nothing found, not extracting anything from AU000L_20250805_112338_303_0090368
2025/08/05 11:23:53-DEBUG-ExtractStars-line:348 - extracted 87 stars from FF_AU000L_20250805_112338_303_0090368.fits
2025/08/05 11:23:53-INFO-DetectStarsAndMeteors-line:231 - Detected stars: 87
2025/08/05 11:23:53-INFO-DetectStarsAndMeteors-line:237 - At least 20 stars, detecting meteors...
2025/08/05 11:23:59-INFO-BufferedCapture-line:1893 - Block's max frame age: 1.028 seconds. Run's dropped frames: 3322
2025/08/05 11:24:00-DEBUG-BufferedCapture-line:1982 - New block of raw frames available for compression with starting time: 1754393028.5505624
2025/08/05 11:24:00-DEBUG-Compression-line:314 - Compressing frame block with start time at: 1754393028.5505624
2025/08/05 11:24:00-DEBUG-BufferedCapture-line:2007 - Created FT file /home/rms/RMS_data/TimeFiles/2025/20250805-217/20250805-217_11/FT_AU000L_20250805_112348.bin for block starting at 1754393028.5505624
2025/08/05 11:24:00-INFO-BufferedCapture-line:1692 - Grabbing a new block of 256 frames...
2025/08/05 11:24:03-INFO-Compression-line:335 - Compression time: 2.774 s
2025/08/05 11:24:07-INFO-Logger-line:266 - rtpjitterbuffer ../gst/rtpmanager/rtpjitterbuffer.c:575:calculate_skew: delta - skew: 0:00:05.424443145 too big, reset skew
2025/08/05 11:24:07-INFO-Compression-line:342 - Saving time: 4.212 s
2025/08/05 11:24:09-DEBUG-DetectionTools-line:74 - Loaded mask: mask.bmp
2025/08/05 11:24:09-INFO-Logger-line:266 - rtpjitterbuffer ../gst/rtpmanager/rtpjitterbuffer.c:575:calculate_skew: delta - skew: 0:00:02.298786398 too big, reset skew
2025/08/05 11:24:09-INFO-DetectionTools-line:78 - Mask is all white, setting it to None.
2025/08/05 11:24:09-DEBUG-Compression-line:359 - Extractor started for: AU000L_20250805_112348_550_0090624
2025/08/05 11:24:10-DEBUG-Compression-line:372 - Added file for detection: FF_AU000L_20250805_112348_550_0090624.fits
2025/08/05 11:24:11-INFO-Logger-line:266 - rtpjitterbuffer ../gst/rtpmanager/rtpjitterbuffer.c:575:calculate_skew: delta - skew: 0:00:01.738694343 too big, reset skew
2025/08/05 11:24:12-INFO-Logger-line:266 - rtpjitterbuffer ../gst/rtpmanager/rtpjitterbuffer.c:575:calculate_skew: delta - skew: 0:00:02.395114533 too big, reset skew

```

And this is the end of the log, it seems as though RMS stopped at this point, unfortunately I do not have any more information.

peschman, 2025-08-08:
I've been running a continuous capture Pi4 on prerelease for several months and have not seen this problem.

Cybis320, 2025-08-13:
Fix is in Prerelease now.

g7gpr, 2025-09-01:
OK, in the interest of tidying up, I'll close this.

closed 2025-09-01 by g7gpr