CroatianMeteorNetwork/RMS #709: RMS_Update seems to only work on CroatianMeteorNetwork as remote

g7gpr, 2025-08-13:
It seems as though RMS_Update only work when pointing at 

```
https://github.com/CroatianMeteorNetwork
```

When running I get this...

```

david@lap-deb-01:~/source/RMS$ Scripts/RMS_Update.sh 
Creating recovery script RMS_Reset.sh...
Recovery script created: Scripts/RMS_Reset.sh
Use './Scripts/RMS_Reset.sh --switch' to return from old branches

====== Checking Git Configuration ======

Valid git repository found
Checking remote configuration...
Found remote 'g7gpr' pointing to: https://github.com/g7gpr/RMS
Found remote 'jt' pointing to: https://github.com/JustinD-T/RMS-ASTRA/
Found remote 'origin' pointing to: https://github.com/CroatianMeteorNetwork/RMS
Found RMS repository at remote 'origin'
Verifying connection to RMS remote...
Successfully connected to RMS repository through 'origin' remote
ERROR: Cannot query remote ref feature/em/radec_report on origin
david@lap-deb-01:~/source/RMS$ Scripts/RMS_Update.sh 
Creating recovery script RMS_Reset.sh...
Recovery script created: Scripts/RMS_Reset.sh
Use './Scripts/RMS_Reset.sh --switch' to return from old branches

====== Checking Git Configuration ======

Valid git repository found
Checking remote configuration...
Found remote 'g7gpr' pointing to: https://github.com/g7gpr/RMS
Found remote 'jt' pointing to: https://github.com/JustinD-T/RMS-ASTRA/
Found remote 'origin' pointing to: https://github.com/CroatianMeteorNetwork/RMS
Found RMS repository at remote 'origin'
Verifying connection to RMS remote...
Successfully connected to RMS repository through 'origin' remote
ERROR: Cannot query remote ref feature/em/radec_report
```

It is looking for my branch, but looking on origin.

Cybis320, 2025-08-13:
Noted. Working on it.

g7gpr, 2025-08-13:
I can no longer contact any of my stations.

However 

```
RMS_REMOTE=$(git for-each-ref --format='%(upstream:remotename)' $(git symbolic-ref -q HEAD))
```

This might be a useful bit of code.